ecog-io/iso15118 #58: SECC Simulator: ModuleNotFoundError: No module named 'aiofile'

RadioX90, 2022-06-01:
First of all, thank you for the iso15118 implementation!

I am trying to run the docker environment on a Ubuntu system. However there seems to be an issue with one of the python packages needed by the SECC simulator, namely "aiofile":
secc_1 | Traceback (most recent call last):
secc_1 |   File "/venv/bin/iso15118", line 5, in <module>
secc_1 |     from iso15118.secc.main import run
secc_1 |   File "/venv/lib/python3.10/site-packages/iso15118/secc/main.py", line 5, in <module>
secc_1 |     from iso15118.secc.controller.simulator import SimEVSEController
secc_1 |   File "/venv/lib/python3.10/site-packages/iso15118/secc/controller/simulator.py", line 12, in <module>
secc_1 |     from aiofile import async_open
secc_1 | ModuleNotFoundError: No module named 'aiofile'

I checked the poetry file etc. and the aiofile packages seems to be in there. But it seems that it is not being installed, although another package e.g. "alabaster" is included:
  • Installing attrs (21.4.0)
  • Installing markupsafe (2.1.1)
  • Installing iniconfig (1.1.1)
  • Updating packaging (20.9 -> 21.3)
  • Installing pluggy (1.0.0)
  • Installing py (1.11.0)
  • Installing pytz (2022.1)
  • Installing alabaster (0.7.12)
  • Installing babel (2.10.1)
  • Installing click (8.1.3)
  • Installing coverage (6.3.3)
  • Installing docutils (0.17.1)
  • Installing imagesize (1.3.0)
  • Installing jinja2 (3.1.2)
  • Installing marshmallow (3.15.0)
  • Installing mccabe (0.6.1)
  • Installing pathspec (0.9.0)
  • Installing pycodestyle (2.8.0)
  • Installing pyflakes (2.4.0)
  • Installing pygments (2.12.0)
  • Installing pytest (7.1.2)
  • Installing python-dateutil (2.8.2)
  • Installing python-dotenv (0.20.0)
  • Installing snowballstemmer (2.2.0)
  • Installing sphinxcontrib-applehelp (1.0.2)
  • Installing sphinxcontrib-devhelp (1.0.2)
  • Installing sphinxcontrib-htmlhelp (2.0.0)
  • Installing sphinxcontrib-jsmath (1.0.1)
  • Installing sphinxcontrib-qthelp (1.0.3)
  • Installing sphinxcontrib-serializinghtml (1.1.5)
  • Installing asynctest (0.13.0)
  • Installing black (22.3.0)
  • Updating cryptography (37.0.2 -> 3.4.6)
  • Installing environs (9.5.0)
  • Installing flake8 (4.0.1)
  • Installing freezegun (1.2.1)
  • Installing isort (5.10.1)
  • Installing psutil (5.9.0)
  • Installing py4j (0.10.9.5)
  • Installing pydantic (1.9.1)
  • Installing pytest-asyncio (0.18.3)
  • Installing pytest-cov (3.0.0)
  • Installing sphinx (4.5.0)

Do you have any idea what is going wrong? Thanks in advance

tropxy, 2022-06-01:
Hi @RadioX90 thanks for the feedback,

This issue was also "reported" here: https://github.com/SwitchEV/iso15118/pull/51
I have a fix for it here: https://github.com/SwitchEV/iso15118/pull/55

However, I believe I actually think that the easiest way is just to install the dev dependencies in the end image as well in docker to solve the issue. I will investigate this, but in the mean time check the comments in the #51 and the code in #55

tropxy, 2022-09-15:
fix #130

closed 2022-09-15 by tropxy