ecog-io/iso15118 #70: Bug in PowerDeliveryReq when sending DCEVPowerDeliveryParameter

tropxy, 2022-06-24:
This bug was in fact raised thanks to the work done by Jörn Schumann from the Fraunhofer Institute.

During one of their tests they found out that when receiving a PowerDeliveryReq message containing the optional `DCEVPowerDeliveryParameter` containing the field `BulkChargingComplete` the iso15118 raised an error.

```python
Decoded dict: {'V2G_Message': {'Header': {'SessionID': '5D61F57AA6225E84'}, 'Body': {'PowerDeliveryReq': {'ChargeProgress': 'Start', 'SAScheduleTupleID': 1, 'ChargingProfile': {'ProfileEntry': [{'ChargingProfileEntryStart': 0, 'ChargingProfileEntryMaxPower': {'Multiplier': 0, 'Unit': 'W', 'Value': 11000}, 'ChargingProfileEntryMaxNumberOfPhasesInUse': 3}, {'ChargingProfileEntryStart': 3600, 'ChargingProfileEntryMaxPower': {'Multiplier': 0, 'Unit': 'W', 'Value': 7000}, 'ChargingProfileEntryMaxNumberOfPhasesInUse': 3}]}, 'DC_EVPowerDeliveryParameter': {'DC_EVStatus': {'EVReady': True, 'EVErrorCode': 'NO_ERROR', 'EVRESSSOC': 49}, 'BulkChargingComplete': False, 'ChargingComplete': False}}}}}

File "pydantic/main.py", line 521, in pydantic.main.BaseModel.parse_obj
File "pydantic/main.py", line 341, in pydantic.main.BaseModel.__init__
pydantic.error_wrappers.ValidationError: 1 validation error for V2GMessage
Body -> PowerDeliveryReq -> DC_EVPowerDeliveryParameter -> BulkChargingComplete extra fields not permitted (type=value_error.extra)
```

The error is due to a typo in here:
https://github.com/SwitchEV/iso15118/blob/master/iso15118/shared/messages/iso15118_2/datatypes.py#L276

Thus, the fix to fix the typo to `alias`.

bachmmmar, 2022-08-31:
This will be fixed with PR #86

martinbachmanndesignwerk, 2022-08-31:
@tropxy Please review PR #86 in order to fix this bug.

closed 2022-09-15 by tropxy