ecog-io/iso15118 #91: The "unitSymbolType" is optional for the DIN70121 protocol, it is rejected now.

henri62, 2022-08-04:
See V2G-DC-297 The PhysicalValueType has 3 attributes: Multiplier, Unit, Value. Here the 'Unit' is optional (in iso spec this is mandatory).
The test now rejects this which is wrong.

See the example log below:
DEBUG    2022-08-04 10:22:40,424 - iso15118.evcc.states.sap_states (142): Chosen protocol: DIN_SPEC_70121
.....
Decoded dict: {'V2G_Message': {'Header': {'SessionID': '4242424242424242'}, 'Body': {'ChargeParameterDiscoveryRes': {'ResponseCode': 'OK', 'EVSEProcessing': 'Finished', 'SAScheduleList': {'SAScheduleTuple': [{'SAScheduleTupleID': 1, 'PMaxSchedule': {'PMaxScheduleID': 1, 'PMaxScheduleEntry': [{'RelativeTimeInterval': {'start': 0, 'duration': 28800}, 'PMax': 32767}]}}]}, 'DC_EVSEChargeParameter': {'DC_EVSEStatus': {'EVSEIsolationStatus': 'Valid', 'EVSEStatusCode': 'EVSE_Ready', 'NotificationMaxDelay': 0, 'EVSENotification': 'None'}, 'EVSEMaximumCurrentLimit': {'Multiplier': 0, 'Value': 180}, 'EVSEMaximumPowerLimit': {'Multiplier': 1, 'Value': 18000}, 'EVSEMaximumVoltageLimit': {'Multiplier': 0, 'Value': 1000}, 'EVSEMinimumCurrentLimit': {'Multiplier': 0, 'Value': 0}, 'EVSEMinimumVoltageLimit': {'Multiplier': 0, 'Value': 200}, 'EVSEPeakCurrentRipple': {'Multiplier': 0, 'Value': 1}}}}}}
INFO    2022-08-04 10:26:13,670 - iso15118.shared.comm_session (371): The data link will Terminate in 2 seconds and the TCP connection will close in 5 seconds. 
INFO    2022-08-04 10:26:13,670 - iso15118.shared.comm_session (375): Reason: EXIDecodingError occurred while processing message  in state ChargeParameterDiscovery:: Error parsing the decoded EXI into a Pydantic class: 6 validation errors for V2GMessage
Body -> ChargeParameterDiscoveryRes -> DC_EVSEChargeParameter -> EVSEMaximumCurrentLimit -> Unit
  field required (type=value_error.missing)
Body -> ChargeParameterDiscoveryRes -> DC_EVSEChargeParameter -> EVSEMaximumPowerLimit -> Unit
  field required (type=value_error.missing)
Body -> ChargeParameterDiscoveryRes -> DC_EVSEChargeParameter -> EVSEMaximumVoltageLimit -> Unit
  field required (type=value_error.missing)
Body -> ChargeParameterDiscoveryRes -> DC_EVSEChargeParameter -> EVSEMinimumCurrentLimit -> Unit
  field required (type=value_error.missing)
Body -> ChargeParameterDiscoveryRes -> DC_EVSEChargeParameter -> EVSEMinimumVoltageLimit -> Unit
  field required (type=value_error.missing)
Body -> ChargeParameterDiscoveryRes -> DC_EVSEChargeParameter -> EVSEPeakCurrentRipple -> Unit
  field required (type=value_error.missing).

MarcMueltin, 2022-09-26:
Hi @henri62 Thanks for making us aware of this, we'll put some more effort into DIN SPEC 70121 testing in the next couple of weeks. Sorry for the delay, we've been extensively working on the ISO 15118 part, specifically making sure Plug and Charge works rock solid, including an integration with the Hubject ecosystem.

shalinnijel2, 2023-01-04:
Hi @henri62 - Thank you for reporting this. I have now created an internal ticket for fixing this issue. Will drop a message here when the fix is available in a release.
AB#3741

closed 2026-10-12 by ikaratass