ecog-io/iso15118 #105: Timeout wrong for the SessionSetup sequence

henri62, 2022-08-15:
The file 'iso15118/shared/messages/timeouts.py' states a value of 20.0 seconds for:
V2G_EVCC_COMMUNICATION_SETUP_TIMEOUT
The specification for iso1 and din however specifies 2.0 seconds.

Am I missing something here or is it simply wrong in this file?

The file 'tests/dinspec/evcc/test_dinspec_evcc_states.py' even has a value of 1 so it looks some (wrong) duplication or mis-alignment is present in the different files?

tropxy, 2022-09-15:
the timeouts of the -20 need to be revised, thanks for checking it

MarcMueltin, 2022-09-26:
The 20 seconds are correct @henri62 @tropxy. You may be confusing this with a timeout for the SessionSetupRes, but what the V2G_EVCC_CommunicationSetup_Timeout is referring to is the maximum time allowed from plugging in the cable until the SessionSetupRes is received (not from sending the SessionSetupReq). 

See Table 111 and requirements in section "8.7.3.2 EVCC Timing for communication session setup"

closed 2022-09-26 by MarcMueltin