ecog-io/iso15118 #114: check_num_e_price_levels - incorrect validation

psz11, 2022-08-23:
[iso15118/shared/messages/iso15118_2/datatypes.py]
Validator assumes that NumEPriceLevels is equal to number of SalesTariffEntries which may not always be the case.

As per ISO15118-2:

NumEPriceLevels:
Defines the overall number of distinct price levels used across all provided SalesTariff elements.

EPriceLevel:
Defines the price level of this SalesTariffEntry (referring to NumEPriceLevels). Small values for the EPriceLevel represent a cheaper TariffEntry. Large values for the EPriceLevel represent a more expensive TariffEntry.

SalesTariffEntry:
Encapsulating element describing all relevant details for one time interval of the SalesTariff. The number of SalesTariffEntry elements is
limited by the parameter MaxEntriesSAScheduleTuple (according to [V2G2-261]). The maximum value shall be 1024.

Export of example config for 'not working' sales tariff:
[SAScheduleList.txt](https://github.com/SwitchEV/iso15118/files/9402711/SAScheduleList.txt)

shalinnijel2, 2023-01-04:
@psz11 Thank you for raising this - and for the example config. :). We will update the validator to reflect this. Will drop you a message when this is fixed.

shalinnijel2, 2023-01-04:
AB#3740

closed 2023-06-19 by shalinnijel2