ecog-io/iso15118 #127: If MO Root is not found it crashes the application

tropxy, 2022-09-07:
```shell
DEBUG    2022-09-07 09:06:35,655 - mqtt_api (303): Sending {"id":"5bde023e-40a9-41bb-8499-80dcf20a4591","type":"update","name":"iso15118_message","data":{"evse_id":"DE*SWT*E123456789","message_type":"Request","message_name":"PaymentServiceSelectionReq","json_payload":{"V2G_Message":{"Body":{"PaymentServiceSelectionReq":{"SelectedPaymentOption":"Contract","SelectedServiceList":{"SelectedService":[{"ServiceID":1}]}}},"Header":{"SessionID":"D24E49E01843440E"}}},"exi_payload":"809802349392780610d1039130001280"}} to topic Topics.ISO15118_OCPP
ERROR    2022-09-07 09:06:35,669 - iso15118.shared.utils (51): [Errno 2] No such file or directory: <CertPath.MO_ROOT_DER: '/extra/iso15118_2/certs/moRootCACert.der'>
Traceback (most recent call last):
  File "iso15118/shared/utils.py", line 49, in wait_for_tasks
  File "iso15118/shared/comm_session.py", line 467, in rcv_loop
  File "iso15118/shared/comm_session.py", line 222, in process_message
  File "iso15118/secc/states/iso15118_2_states.py", line 499, in process_message
  File "iso15118/secc/states/iso15118_2_states.py", line 879, in process_message
  File "iso15118/shared/security.py", line 333, in load_cert
FileNotFoundError: [Errno 2] No such file or directory: <CertPath.MO_ROOT_DER: '/extra/iso15118_2/certs/moRootCACert.der'>
DEBUG    2022-09-07 09:06:35,741 - mqtt_api (31): Trying to open the path: /extra/iso15118_service/mqtt_api/schemas/Iso15118MessageUpdate.json
DEBUG    2022-09-07 09:06:35,849 - mqtt_api (303): Sending {"id":"cf6eb61a-1272-43b1-a1b2-dfa738634280","type":"update","name":"iso15118_message","data":{"evse_id":"DE*SWT*E123456789","message_type":"Response","message_name":"PaymentServiceSelectionRes","json_payload":{"V2G_Message":{"Body":{"PaymentServiceSelectionRes":{"ResponseCode":"OK"}},"Header":{"SessionID":"D24E49E01843440E"}}},"exi_payload":"809802349392780610d103914000"}} to topic Topics.ISO15118_OCPP
```

The MO root is not mandatory to be used to verify the certificates as we may not have it, so we should try to get the file, but if is not available, we shall move on and proceed

tropxy, 2022-09-07:
full log
[test_mo_root_not_found.txt.zip](https://github.com/SwitchEV/iso15118/files/9505884/test_mo_root_not_found.txt.zip)

shalinnijel2, 2022-09-07:
I guess something went wrong with the attachment upload - it says "not found" when I click on it...

tropxy, 2022-09-08:
[test_mo_root_not_found.txt](https://github.com/SwitchEV/iso15118/files/9526848/test_mo_root_not_found.txt)

tropxy, 2022-09-08:
> I guess something went wrong with the attachment upload - it says "not found" when I click on it...

now should work ;)

closed 2022-09-15 by tropxy