ecog-io/iso15118 #142: ACEVSEStatus is static in ChargingStatusRes

tropxy, 2022-10-05:
```python
charging_status_res = ChargingStatusRes(
            response_code=ResponseCode.OK,
            evse_id=await self.comm_session.evse_controller.get_evse_id(
                Protocol.ISO_15118_2
            ),
            sa_schedule_tuple_id=self.comm_session.selected_schedule,
            ac_evse_status=ACEVSEStatus(
                notification_max_delay=0,
                evse_notification=EVSENotification.NONE,
                rcd=False,
            ),
            # TODO Could maybe request an OCPP setting that determines
            #      whether or not a receipt is required and when
            #      (probably only makes sense at the beginning and end of
            #      a charging session). If true, set MeterInfo.
            # meter_info=await self.comm_session.evse_controller.get_meter_info_v2(),
            receipt_required=False,
        )
```

shalinnijel2, 2022-10-16:
This is fixed under [146](https://github.com/SwitchEV/iso15118/commit/15c3a5ad456c104aa1ef11edd82051f2b9264e98). Leaving this open until it becomes available in a release.

tropxy, 2022-10-25:
> This is fixed under [146](https://github.com/SwitchEV/iso15118/commit/15c3a5ad456c104aa1ef11edd82051f2b9264e98). Leaving this open until it becomes available in a release.

The fix was merged, so I will close it. If a bug still emerges later on related to this we can reopen or create a new issue

closed 2022-10-25 by tropxy