ecog-io/iso15118 #152: How do I run -20 in this project?

GUANMINLIAO, 2022-10-22:
Using the step causes to run -2
but I need to watch the messages flow in -20

thanks you

shalinnijel2, 2022-10-22:
In the .env file, set the following line (or modify as follows) :

```
PROTOCOLS=ISO_15118_20_AC
```

All the other possible options supported by the .env file are described here:
https://github.com/SwitchEV/iso15118#environment-settings

GUANMINLIAO, 2022-10-24:
Hello,but i seem encountering some problem in certificate installation problem,it is my fault?
Thank you!
EVCC:
`(iso15118-py3.10) iso15118@iso15118-LIFEBOOK-SH572:~/桌面/iso15118-master$ make run-evcc
make[1]: 进入目录“/home/iso15118/桌面/iso15118-master”
/home/iso15118/.cache/pypoetry/virtualenvs/iso15118-j4EfOCiL-py3.10/bin/python iso15118/evcc/main.py
DEBUG    2022-10-24 16:14:22,782 - asyncio (54): Using selector: EpollSelector
INFO    2022-10-24 16:14:24,397 - iso15118.evcc.evcc_settings (93): EVCC settings:
INFO    2022-10-24 16:14:24,397 - iso15118.evcc.evcc_settings (95): PKI_PATH                      : /home/iso15118/桌面/iso15118-master/iso15118/shared/pki
INFO    2022-10-24 16:14:24,397 - iso15118.evcc.evcc_settings (95): MESSAGE_LOG_EXI               : False
INFO    2022-10-24 16:14:24,397 - iso15118.evcc.evcc_settings (95): V20_SERVICE_CONFIG            : /home/iso15118/桌面/iso15118-master/iso15118/shared/examples/15118_20_evse_service_config.json
INFO    2022-10-24 16:14:24,398 - iso15118.evcc.evcc_settings (95): MESSAGE_LOG_JSON              : True
INFO    2022-10-24 16:14:24,398 - iso15118.evcc.evcc_settings (95): ENABLE_TLS_1_3                : False
INFO    2022-10-24 16:14:24,398 - iso15118.evcc.evcc_settings (97): MAX_SUPPORTING_POINTS         : 1024
INFO    2022-10-24 16:14:24,399 - iso15118.evcc.evcc_settings (97): NETWORK_INTERFACE             : wlp11s0
INFO    2022-10-24 16:14:24,399 - iso15118.evcc.evcc_settings (97): EVCC_ENFORCE_TLS              : False
INFO    2022-10-24 16:14:24,399 - iso15118.evcc.evcc_settings (97): MAX_CONTRACT_CERTS            : 3
INFO    2022-10-24 16:14:24,399 - iso15118.evcc.evcc_settings (97): SDP_RETRY_CYCLES              : 1
INFO    2022-10-24 16:14:24,399 - iso15118.evcc.evcc_settings (97): LOG_LEVEL                     : DEBUG
INFO    2022-10-24 16:14:24,399 - iso15118.evcc.evcc_settings (97): EVCC_USE_TLS                  : False
INFO    2022-10-24 16:14:24,401 - iso15118.shared.exi_codec (174): EXI Codec version: 1.55
INFO    2022-10-24 16:14:24,402 - iso15118.evcc (28): Starting 15118 version: 0.12.0
INFO    2022-10-24 16:14:24,402 - iso15118.evcc.comm_session_handler (295): Communication session handler started
DEBUG    2022-10-24 16:14:24,402 - iso15118.evcc.comm_session_handler (374): Initiating new SDP cycle, 0 more cycles(s) left
INFO    2022-10-24 16:14:24,403 - iso15118.evcc.comm_session_handler (380): Remaining SDP requests: 50
DEBUG    2022-10-24 16:14:24,403 - iso15118.evcc.transport.udp_client (90): UDP client socket ready
INFO    2022-10-24 16:14:24,504 - iso15118.evcc.comm_session_handler (317): Sending SDPRequest: [Security: NO_TLS, Protocol: TCP]
DEBUG    2022-10-24 16:14:24,504 - iso15118.evcc.transport.udp_client (170): Message sent: [Header = [0x1, 0xfe, 0x9000, 2], Payload = 1000)]
INFO    2022-10-24 16:14:24,664 - iso15118.evcc.transport.udp_client (105): Received datagram from UDP server at address ('fe80::c1b5:bb3d:287a:db0f', 15118, 0, 3)
INFO    2022-10-24 16:14:24,665 - iso15118.evcc.comm_session_handler (462): SDPResponse received: [ IP address: fe80::c1b5:bb3d:287a:db0f, Port: 60897 , Security: NO_TLS , Transport: TCP ]
INFO    2022-10-24 16:14:24,665 - iso15118.evcc.comm_session_handler (398): Starting TCP client, trying to connect to fe80::c1b5:bb3d:287a:db0f at port 60897 ...
INFO    2022-10-24 16:14:24,667 - iso15118.evcc.transport.udp_client (105): Received datagram from UDP server at address ('fe80::c1b5:bb3d:287a:db0f', 15118, 0, 3)
INFO    2022-10-24 16:14:24,682 - iso15118.evcc.comm_session_handler (405): TCP client connected
INFO    2022-10-24 16:14:24,693 - iso15118.shared.comm_session (341): Starting a new communication session
INFO    2022-10-24 16:14:24,693 - iso15118.shared.states (137): Entered state SupportedAppProtocol
DEBUG    2022-10-24 16:14:24,693 - iso15118.shared.states (141): Waiting for up to 2.0 s
INFO    2022-10-24 16:14:24,694 - iso15118.shared.exi_codec (245): Message to encode (ns=urn:iso:15118:2:2010:AppProtocol): {"supportedAppProtocolReq": {"AppProtocol": [{"ProtocolNamespace": "urn:iso:15118:2:2013:MsgDef", "VersionNumberMajor": 2, "VersionNumberMinor": 0, "SchemaID": 1, "Priority": 1}, {"ProtocolNamespace": "urn:iso:std:iso:15118:-20:AC", "VersionNumberMajor": 1, "VersionNumberMinor": 0, "SchemaID": 2, "Priority": 2}, {"ProtocolNamespace": "urn:din:70121:2012:MsgDef", "VersionNumberMajor": 2, "VersionNumberMinor": 0, "SchemaID": 3, "Priority": 3}]}}
INFO    2022-10-24 16:14:24,754 - iso15118.shared.comm_session (415): Sending supportedAppProtocolReq
INFO    2022-10-24 16:14:24,960 - iso15118.shared.exi_codec (299): Decoded message (ns=urn:iso:15118:2:2010:AppProtocol): {"supportedAppProtocolRes":{"ResponseCode":"OK_SuccessfulNegotiation","SchemaID":2}}
INFO    2022-10-24 16:14:24,961 - iso15118.shared.comm_session (231): supportedAppProtocolRes received
INFO    2022-10-24 16:14:24,961 - iso15118.evcc.states.sap_states (142): Chosen protocol: ISO_15118_20_AC
INFO    2022-10-24 16:14:24,962 - iso15118.shared.exi_codec (245): Message to encode (ns=urn:iso:std:iso:15118:-20:CommonMessages): {"SessionSetupReq": {"Header": {"SessionID": "00", "TimeStamp": 1666599264}, "EVCCID": "WMIV1234567890ABCDEX"}}
INFO    2022-10-24 16:14:25,254 - iso15118.shared.comm_session (415): Sending SessionSetupReq
INFO    2022-10-24 16:14:25,254 - iso15118.shared.states (137): Entered state SessionSetup
DEBUG    2022-10-24 16:14:25,254 - iso15118.shared.states (141): Waiting for up to 20.0 s
[... cut ...]
  File "/home/liao/桌面/iso15118-master/iso15118/secc/states/iso15118_20_states.py", line 296, in process_message
    raise NotImplementedError("CertificateInstallation not yet implemented")
NotImplementedError: CertificateInstallation not yet implemented
^CWARNING    2022-10-24 16:15:10,514 - iso15118.secc.transport.tcp_server (122): Closing TCP server
WARNING    2022-10-24 16:15:10,514 - iso15118.secc.transport.tcp_server (122): Closing TCP server
DEBUG    2022-10-24 16:15:10,515 - __main__ (26): SECC program terminated manually
`

closed 2022-10-25 by GUANMINLIAO

tropxy, 2022-10-25:
Hi @GUANMINLIAO , I got a notification that initially you had some issue with the certificates. I believe you have by chance found a bug in the -20: inspecting the logs you have shared, seems that after a NO_TLS communication has been established, we still offered PnC in the `AuthorizationSetupResponse` message: `{"AuthorizationSetupRes": {"Header": {"SessionID": "3A25C9B9A5FA6D36", "TimeStamp": 1666599266}, "ResponseCode": "OK", "AuthorizationServices": ["PnC", "EIM"],...`

This is incorrect and we shall treat it as a bug, so I will open an issue for this. Thank you!