ecog-io/iso15118 #153: PnC is offered in ISO 15118-20 during a NO_TLS connection

tropxy, 2022-10-25:
User  @GUANMINLIAO reported in here #152 a problem on setting up a -20 session. 

`EVCC`:
```shell
EVCC:
(iso15118-py3.10) iso15118@iso15118-LIFEBOOK-SH572:~/桌面/iso15118-master$ make run-evcc make[1]: 进入目录“/home/iso15118/桌面/iso15118-master” /home/iso15118/.cache/pypoetry/virtualenvs/iso15118-j4EfOCiL-py3.10/bin/python iso15118/evcc/main.py DEBUG 2022-10-24 16:14:22,782 - asyncio (54): Using selector: EpollSelector INFO 2022-10-24 16:14:24,397 - iso15118.evcc.evcc_settings (93): EVCC settings: INFO 2022-10-24 16:14:24,397 - iso15118.evcc.evcc_settings (95): PKI_PATH : /home/iso15118/桌面/iso15118-master/iso15118/shared/pki INFO 2022-10-24 16:14:24,397 - iso15118.evcc.evcc_settings (95): MESSAGE_LOG_EXI : False INFO 2022-10-24 16:14:24,397 - iso15118.evcc.evcc_settings (95): V20_SERVICE_CONFIG : /home/iso15118/桌面/iso15118-master/iso15118/shared/examples/15118_20_evse_service_config.json INFO 2022-10-24 16:14:24,398 - iso15118.evcc.evcc_settings (95): MESSAGE_LOG_JSON : True INFO 2022-10-24 16:14:24,398 - iso15118.evcc.evcc_settings (95): ENABLE_TLS_1_3 : False INFO 2022-10-24 16:14:24,398 - iso15118.evcc.evcc_settings (97): MAX_SUPPORTING_POINTS : 1024 INFO 2022-10-24 16:14:24,399 - iso15118.evcc.evcc_settings (97): NETWORK_INTERFACE : wlp11s0 INFO 2022-10-24 16:14:24,399 - iso15118.evcc.evcc_settings (97): EVCC_ENFORCE_TLS : False INFO 2022-10-24 16:14:24,399 - iso15118.evcc.evcc_settings (97): MAX_CONTRACT_CERTS : 3 INFO 2022-10-24 16:14:24,399 - iso15118.evcc.evcc_settings (97): SDP_RETRY_CYCLES : 1 INFO 2022-10-24 16:14:24,399 - iso15118.evcc.evcc_settings (97): LOG_LEVEL : DEBUG INFO 2022-10-24 16:14:24,399 - iso15118.evcc.evcc_settings (97): EVCC_USE_TLS : False INFO 2022-10-24 16:14:24,401 - iso15118.shared.exi_codec (174): EXI Codec version: 1.55 INFO 2022-10-24 16:14:24,402 - iso15118.evcc (28): Starting 15118 version: 0.12.0 INFO 2022-10-24 16:14:24,402 - iso15118.evcc.comm_session_handler (295): Communication session handler started DEBUG 2022-10-24 16:14:24,402 - iso15118.evcc.comm_session_handler (374): Initiating new SDP cycle, 0 more cycles(s) left INFO 2022-10-24 16:14:24,403 - iso15118.evcc.comm_session_handler (380): Remaining SDP requests: 50 DEBUG 2022-10-24 16:14:24,403 - iso15118.evcc.transport.udp_client (90): UDP client socket ready INFO 2022-10-24 16:14:24,504 - iso15118.evcc.comm_session_handler (317): Sending SDPRequest: [Security: NO_TLS, Protocol: TCP] DEBUG 2022-10-24 16:14:24,504 - iso15118.evcc.transport.udp_client (170): Message sent: [Header = [0x1, 0xfe, 0x9000, 2], Payload = 1000)] INFO 2022-10-24 16:14:24,664 - iso15118.evcc.transport.udp_client (105): Received datagram from UDP server at address ('fe80::c1b5:bb3d:287a:db0f', 15118, 0, 3) INFO 2022-10-24 16:14:24,665 - iso15118.evcc.comm_session_handler (462): SDPResponse received: [ IP address: fe80::c1b5:bb3d:287a:db0f, Port: 60897 , Security: NO_TLS , Transport: TCP ] INFO 2022-10-24 16:14:24,665 - iso15118.evcc.comm_session_handler (398): Starting TCP client, trying to connect to fe80::c1b5:bb3d:287a:db0f at port 60897 ... INFO 2022-10-24 16:14:24,667 - iso15118.evcc.transport.udp_client (105): Received datagram from UDP server at address ('fe80::c1b5:bb3d:287a:db0f', 15118, 0, 3) INFO 2022-10-24 16:14:24,682 - iso15118.evcc.comm_session_handler (405): TCP client connected INFO 2022-10-24 16:14:24,693 - iso15118.shared.comm_session (341): Starting a new communication session INFO 2022-10-24 16:14:24,693 - iso15118.shared.states (137): Entered state SupportedAppProtocol DEBUG 2022-10-24 16:14:24,693 - iso15118.shared.states (141): Waiting for up to 2.0 s INFO 2022-10-24 16:14:24,694 - iso15118.shared.exi_codec (245): Message to encode (ns=urn:iso:15118:2:2010:AppProtocol): {"supportedAppProtocolReq": {"AppProtocol": [{"ProtocolNamespace": "urn:iso:15118:2:2013:MsgDef", "VersionNumberMajor": 2, "VersionNumberMinor": 0, "SchemaID": 1, "Priority": 1}, {"ProtocolNamespace": "urn:iso:std:iso:15118:-20:AC", "VersionNumberMajor": 1, "VersionNumberMinor": 0, "SchemaID": 2, "Priority": 2}, {"ProtocolNamespace": "urn:din:70121:2012:MsgDef", "VersionNumberMajor": 2, "VersionNumberMinor": 0, "SchemaID": 3, "Priority": 3}]}} INFO 2022-10-24 16:14:24,754 - iso15118.shared.comm_session (415): Sending supportedAppProtocolReq INFO 2022-10-24 16:14:24,960 - iso15118.shared.exi_codec (299): Decoded message (ns=urn:iso:15118:2:2010:AppProtocol): {"supportedAppProtocolRes":{"ResponseCode":"OK_SuccessfulNegotiation","SchemaID":2}} INFO 2022-10-24 16:14:24,961 - iso15118.shared.comm_session (231): supportedAppProtocolRes received INFO 2022-10-24 16:14:24,961 - iso15118.evcc.states.sap_states (142): Chosen protocol: ISO_15118_20_AC INFO 2022-10-24 16:14:24,962 - iso15118.shared.exi_codec (245): Message to encode (ns=urn:iso:std:iso:15118:-20:CommonMessages): {"SessionSetupReq": {"Header": {"SessionID": "00", "TimeStamp": 1666599264}, "EVCCID": "WMIV1234567890ABCDEX"}} INFO 2022-10-24 16:14:25,254 - iso15118.shared.comm_session (415): Sending SessionSetupReq INFO 2022-10-24 16:14:25,254 - iso15118.shared.states (137): Entered state SessionSetup DEBUG 2022-10-24 16:14:25,254 - iso15118.shared.states (141): Waiting for up to 20.0 s INFO 2022-10-24 16:14:25,733 - iso15118.shared.exi_codec (299): Decoded message (ns=urn:iso:std:iso:15118:-20:CommonMessages): {"SessionSetupRes":{"Header":{"SessionID":"3A25C9B9A5FA6D36","TimeStamp":1666599265},"ResponseCode":"OK_NewSessionEstablished","EVSEID":"UK123E1234"}} INFO 2022-10-24 16:14:25,733 - iso15118.shared.comm_session (231): SessionSetupRes received INFO 2022-10-24 16:14:25,734 - iso15118.shared.exi_codec (245): Message to encode (ns=urn:iso:std:iso:15118:-20:CommonMessages): {"AuthorizationSetupReq": {"Header": {"SessionID": "3A25C9B9A5FA6D36", "TimeStamp": 1666599265}}} INFO 2022-10-24 16:14:25,874 - iso15118.shared.comm_session (415): Sending AuthorizationSetupReq INFO 2022-10-24 16:14:25,874 - iso15118.shared.states (137): Entered state AuthorizationSetup DEBUG 2022-10-24 16:14:25,874 - iso15118.shared.states (141): Waiting for up to 2.0 s INFO 2022-10-24 16:14:26,814 - iso15118.shared.exi_codec (299): Decoded message (ns=urn:iso:std:iso:15118:-20:CommonMessages): {"AuthorizationSetupRes":{"Header":{"SessionID":"3A25C9B9A5FA6D36","TimeStamp":1666599266},"ResponseCode":"OK","AuthorizationServices":["PnC","EIM"],"CertificateInstallationService":true,"PnC_ASResAuthorizationMode":{"GenChallenge":"xvYFfT3HTlVeXp/Uu+O33A=="}}} INFO 2022-10-24 16:14:26,815 - iso15118.shared.comm_session (231): AuthorizationSetupRes received INFO 2022-10-24 16:14:26,815 - iso15118.shared.exi_codec (245): Message to encode (ns=urn:iso:std:iso:15118:-20:CommonMessages): {"OEMProvisioningCertificateChain": {"Id": "id1", "Certificate": "MIIB4DCCAYagAwIBAgICMEAwCgYIKoZIzj0EAwIwRzESMBAGA1UEAwwJT0VNU3ViQ0EyMQ8wDQYDVQQKDAZTd2l0Y2gxCzAJBgNVBAYTAlVLMRMwEQYKCZImiZPyLGQBGRYDT0VNMB4XDTIyMTAxMzA4NTMwMVoXDTI2MTAxMjA4NTMwMVowSTEUMBIGA1UEAwwLT0VNUHJvdkNlcnQxDzANBgNVBAoMBlN3aXRjaDELMAkGA1UEBhMCVUsxEzARBgoJkiaJk/IsZAEZFgNPRU0wWTATBgcqhkjOPQIBBggqhkjOPQMBBwNCAAT3olr3o61LlbhF4Wl2r5Z62miu5xaw3xeM4/TM4E4qyPs4QPVOm/htowNQSYFRwWn2nPr4kGo6JSbfsUiGarQeo2AwXjAMBgNVHRMBAf8EAjAAMA4GA1UdDwEB/wQEAwIDiDAdBgNVHQ4EFgQU22xt8CJzu3fRRw3NFc+Y0AbuRF4wHwYDVR0jBBgwFoAUCKiN5SlNVdG2nrDtc0gITadmtiAwCgYIKoZIzj0EAwIDSAAwRQIhAJx2iuDNGE++1JaHqLir03sTXzKLRfPwuiiJz1FzBjvAAiB9qNeRPkQyYiWgtBxxTmzpUvsYl+PiFcdlqKhV7sPNZg==", "SubCertificates": {"Certificate": ["MIIB5TCCAYqgAwIBAgICMD8wCgYIKoZIzj0EAwIwRzESMBAGA1UEAwwJT0VNU3ViQ0ExMQ8wDQYDVQQKDAZTd2l0Y2gxCzAJBgNVBAYTAlVLMRMwEQYKCZImiZPyLGQBGRYDT0VNMB4XDTIyMTAxMzA4NTMwMVoXDTI2MTAxMjA4NTMwMVowRzESMBAGA1UEAwwJT0VNU3ViQ0EyMQ8wDQYDVQQKDAZTd2l0Y2gxCzAJBgNVBAYTAlVLMRMwEQYKCZImiZPyLGQBGRYDT0VNMFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAE+C8ddo1UH82FPI7diX8BySsMU8CJENnCJAMDDPmsWY/Mb8T97AOUdCIohSMkNOVi3+oxqRFlKIUSyb96KEKuZqNmMGQwEgYDVR0TAQH/BAgwBgEB/wIBADAOBgNVHQ8BAf8EBAMCAQYwHQYDVR0OBBYEFAiojeUpTVXRtp6w7XNICE2nZrYgMB8GA1UdIwQYMBaAFLSs5N2k5VDWXSab4GJHzGdi8pfBMAoGCCqGSM49BAMCA0kAMEYCIQDJYjp0MOuH2yWuvpig3NrSTefRGS7fIuyA2hps647GbwIhAOWRdpbEltPt2+Y8ArH51k5Ej1+aW74Lus2lmtO50r23", "MIIB5DCCAYqgAwIBAgICMD4wCgYIKoZIzj0EAwIwRzESMBAGA1UEAwwJT0VNUm9vdENBMQ8wDQYDVQQKDAZTd2l0Y2gxCzAJBgNVBAYTAlVLMRMwEQYKCZImiZPyLGQBGRYDT0VNMB4XDTIyMTAxMzA4NTMwMVoXDTI2MTAxMjA4NTMwMVowRzESMBAGA1UEAwwJT0VNU3ViQ0ExMQ8wDQYDVQQKDAZTd2l0Y2gxCzAJBgNVBAYTAlVLMRMwEQYKCZImiZPyLGQBGRYDT0VNMFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAE5tUxWVfTbIhQqLFqiLjRFsXw06IdKaJywUcW4JJCJ9EXzGOef4vIc0VORg5uEelVrO9KQX0kFMlsHWOdVV8vbKNmMGQwEgYDVR0T [... cut ...] ver DEBUG 2022-10-24 16:15:10,515 - __main__ (26): SECC program terminated manually
```

After inspecting the trench of logs provided, I concluded that we are offering that after a NO_TLS communication has been established, PnC as an option in the `AuthorizationSetupResponse` message: `{"AuthorizationSetupRes": {"Header": {"SessionID": "3A25C9B9A5FA6D36", "TimeStamp": 1666599266}, "ResponseCode": "OK", "AuthorizationServices": ["PnC", "EIM"],...`

This is incorrect and we shall treat it as a bug

GUANMINLIAO, 2022-10-29:
@tropxy 
Thank you for mentioning it.

shalinnijel2, 2023-01-11:
https://dev.azure.com/switch-ev/Josev/_sprints/taskboard/Josev%20Team/Josev/Sprint%2039?workitem=3801