ecog-io/iso15118 #156: Raspberry Pi?

hughsheehy, 2022-11-07:
Trying to install this on a Pi4. 
Various issues occur, some of which I've gotten past. 
First, lots of errors related to openssl. That seems to be fixable by installing libssl-dev
Then we start getting lots of errors around wheels. Oddly, the specific libraries seem to install ok in the base Python environment, but not in the virtual environment that Poetry seems to create. 

Note - I've done several installs on "normal" Linux. No problems. WOrks fine. But on a Pi? Am on day 2.5 trying to get it to work. 

Any previous experience

H

shalinnijel2, 2022-11-08:
Hi, thank you for reporting the issue. Could you elaborate a bit on the issues you are facing at the moment. I'm hoping to look at an issue reported [here](https://github.com/SwitchEV/iso15118/issues/157) soon which was on a Pi.

hughsheehy, 2022-11-08:
I didn't take good notes, sadly. 
First issue (IIRC) was that there were cryptography and opensll errors when I did the install of Poetry. That seemed to be fixable by installing libssl-dev. There were then a variety (sorry) of errors in the Poetry install that I sorted by fishing around stackexchange for generic python wheels/library/pip errors.  Tedious, and I didn't take notes. 

Then, once the install of Poetry worked, I tried make install-local. That resulted in errors on cffi and ffi. 

I've just solved that by installing libffi-dev. 

And now when i try make install-local I get  cryptography errors again.  The main install ran and installed cffi, then a bunch more (see below) then broke on toml? I think it's breaking on toml. The install messages are below and then the error message is further down. 
  _• Installing environs (9.5.0)
  • Installing flake8 (4.0.1)
  • Installing freezegun (1.2.2)
  • Installing isort (5.10.1)
  • Installing mypy (0.982)
  • Installing psutil (5.9.2)
  • Installing py4j (0.10.9.7)
  • Installing pydantic (1.10.2)
  • Installing pytest-asyncio (0.18.3)
  • Installing pytest-cov (3.0.0)
  • Installing toml (0.10.2)
make: *** [Makefile:76: install-local] Error 1_

And then here's the error message. 

_Command ['/home/goeve/.cache/pypoetry/virtualenvs/iso15118-nja0e1FM-py3.9/bin/python', '-m', 'pip', 'install', '--use-pep517', '--disable-pip-version-check', '--prefix', '/home/goeve/.cache/pypoetry/virtualenvs/iso15118-nja0e1FM-py3.9', '--no-deps', '/home/goeve/.cache/pypoetry/artifacts/bd/b4/aa/4b3d4a0414e0fcf9b59b7188c51e64e59883969fe6b783d32de38ad627/cryptography-38.0.1.tar.gz'] errored with the following return code 1, and output: 
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
  Processing /home/goeve/.cache/pypoetry/artifacts/bd/b4/aa/4b3d4a0414e0fcf9b59b7188c51e64e59883969fe6b783d32de38ad627/cryptography-38.0.1.tar.gz
    Installing build dependencies: started
    Installing build dependencies: finished with status 'done'
    Getting requirements to build wheel: started
    Getting requirements to build wheel: finished with status 'done'
    Preparing metadata (pyproject.toml): started
    Preparing metadata (pyproject.toml): finished with status 'done'
  Building wheels for collected packages: cryptography
    Building wheel for cryptography (pyproject.toml): started
    Building wheel for cryptography (pyproject.toml): finished with status 'canceled'
  ERROR: Operation cancelled by user_

hughsheehy, 2022-11-08:
Here's more of the error message

 _at ~/.local/share/pypoetry/venv/lib/python3.9/site-packages/poetry/utils/env.py:1476 in _run
      1472│                 output = subprocess.check_output(
      1473│                     command, stderr=subprocess.STDOUT, env=env, **kwargs
      1474│                 )
      1475│         except CalledProcessError as e:
    → 1476│             raise EnvCommandError(e, input=input_)
      1477│ 
      1478│         return decode(output)
      1479│ 
      1480│     def execute(self, bin: str, *args: str, **kwargs: Any) -> int:

The following error occurred when trying to handle this error:

  PoetryException

  Failed to install /home/goeve/.cache/pypoetry/artifacts/bd/b4/aa/4b3d4a0414e0fcf9b59b7188c51e64e59883969fe6b783d32de38ad627/cryptography-38.0.1.tar.gz

  at ~/.local/share/pypoetry/venv/lib/python3.9/site-packages/poetry/utils/pip.py:51 in pip_install
       47│ 
       48│     try:
       49│         return environment.run_pip(*args)
       50│     except EnvCommandError as e:
    →  51│         raise PoetryException(f"Failed to install {path.as_posix()}") from e
       52│_

hughsheehy, 2022-11-08:
**And if I to into that cache directory and unzip then run the files in the cryptography tar file,  I get this error.** 

_goeve@raspberrypiflet:~/.cache/pypoetry/artifacts/bd/b4/aa/4b3d4a0414e0fcf9b59b7188c51e64e59883969fe6b783d32de38ad627/cryptography-38.0.1 $ python3 setup.py 

        =============================DEBUG ASSISTANCE==========================
        If you are seeing an error here please try the following to
        successfully install cryptography:

        Upgrade to the latest pip and try again. This will fix errors for most
        users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
        =============================DEBUG ASSISTANCE==========================
        
Traceback (most recent call last):
  File "/home/goeve/.cache/pypoetry/artifacts/bd/b4/aa/4b3d4a0414e0fcf9b59b7188c51e64e59883969fe6b783d32de38ad627/cryptography-38.0.1/setup.py", line 17, in <module>
    from setuptools_rust import RustExtension_

hughsheehy, 2022-11-08:
so then did

pip install setuptools_rust

and now it works. 

Installed and running.

closed 2022-11-23 by shalinnijel2