ecog-io/iso15118 #166: EVCC AC charge

macellari-mac, 2022-11-23:
Hi,
I'm testing iso15118 EVCC but it seems that it only works in DC mode.
I cannot find documentation explaining how to set the EVCC to work in AC mode.
If I setup the SECC to work only in AC mode then the EVCC stops in early phases.
If I setup the SECC to work in both AC/DC mode then the EVCC simulator goes on in DC mode.

Can you help please?
Thanks
Lucio

shalinnijel2, 2022-11-23:
Hi,
The EVCC simulator picks up it's desired energy transfer mode from this method: https://github.com/SwitchEV/iso15118/blob/7e4c16d865cd0214dcbb0c6595b20d77aca1e977/iso15118/evcc/controller/simulator.py#L161

If you wish to switch to AC, you could change the method to return EnergyTransferModeEnum.AC_THREE_PHASE_CORE

macellari-mac, 2022-11-23:
Thank you for your help.
Can I suggest to add this mode setting to .env ?

closed 2022-11-23 by macellari-mac