ecog-io/iso15118 #167: Possible to install without Poetry?

hughsheehy, 2022-11-27:
Hi

As recently, i did finally get this rather excellent library installed on a Raspberry Pi, though it took quite a while to get Poetry installed. On a BeagleBone Black, that's turning out to be even more difficult. https://github.com/python-poetry/poetry/issues/7108

Is there a way of installing this library without Poetry? 

H

tropxy, 2022-11-30:
Hi @hughsheehy, yes, you can install without poetry if you instead do `pip install .` in the root of the project where the pyproject.toml is (dont forget the "." after install). Let me know if it works

hughsheehy, 2022-11-30:
Thanks!

I will try. I did get poetry working, but on a small platform like a Pi or (even more) like a Beagle Bone, Poetry is a major install!  Takes HOURS. And requires lots of libraries that don't come on a Pi/Beagle.

H

hughsheehy, 2022-11-30:
Not sure I did the right sequence of commands. 

Did
- pip install .
- make run-secc

Got this. Fails for lack of aoifile...down at the bottom. 

++++++++++++++++++++++++++++++++++++++++++++++++

debian@BeagleBone:~/iso15118-master$ pip install .
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/debian/iso15118-master
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting environs<10.0.0,>=9.5.0
  Downloading https://www.piwheels.org/simple/environs/environs-9.5.0-py2.py3-none-any.whl (12 kB)
Collecting py4j<0.11.0,>=0.10.9
  Downloading https://www.piwheels.org/simple/py4j/py4j-0.10.9.7-py2.py3-none-any.whl (200 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 200.5/200.5 kB 928.9 kB/s eta 0:00:00
Collecting pydantic<2.0.0,>=1.9.0
  Downloading https://www.piwheels.org/simple/pydantic/pydantic-1.10.2-cp39-cp39-linux_armv7l.whl (12.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 12.7/12.7 MB 1.2 MB/s eta 0:00:00
Collecting cryptography==38.0.1
  Downloading https://www.piwheels.org/simple/cryptography/cryptography-38.0.1-cp39-cp39-linux_armv7l.whl (2.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.5/2.5 MB 838.9 kB/s eta 0:00:00
Collecting psutil<6.0.0,>=5.9.0
  Downloading https://www.piwheels.org/simple/psutil/psutil-5.9.4-cp39-abi3-linux_armv7l.whl (274 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 274.2/274.2 kB 657.3 kB/s eta 0:00:00
Requirement already satisfied: cffi>=1.12 in /home/debian/.local/lib/python3.9/site-packages (from cryptography==38.0.1->iso15118==0.14.1) (1.15.1)
Collecting marshmallow>=3.0.0
  Downloading marshmallow-3.19.0-py3-none-any.whl (49 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 49.1/49.1 kB 705.5 kB/s eta 0:00:00
Collecting python-dotenv
  Downloading https://www.piwheels.org/simple/python-dotenv/python_dotenv-0.21.0-py3-none-any.whl (18 kB)
Requirement already satisfied: typing-extensions>=4.1.0 in /usr/local/lib/python3.9/dist-packages (from pydantic<2.0.0,>=1.9.0->iso15118==0.14.1) (4.4.0)
Requirement already satisfied: pycparser in /home/debian/.local/lib/python3.9/site-packages (from cffi>=1.12->cryptography==38.0.1->iso15118==0.14.1) (2.21)
Collecting packaging>=17.0
  Using cached https://www.piwheels.org/simple/packaging/packaging-21.3-py3-none-any.whl (40 kB)
Collecting pyparsing!=3.0.5,>=2.0.2
  Using cached https://www.piwheels.org/simple/pyparsing/pyparsing-3.0.9-py3-none-any.whl (98 kB)
Building wheels for collected packages: iso15118
  Building wheel for iso15118 (pyproject.toml) ... done
  Created wheel for iso15118: filename=iso15118-0.14.1-py3-none-any.whl size=11953618 sha256=83a68478764c3628a0a750c526b515b38a728c0d32ccddddbb68e1de571c7d87
  Stored in directory: /home/debian/.cache/pip/wheels/63/7b/61/18139573307d3228a34de86725f37b85de7883887a5a25356c
Successfully built iso15118
Installing collected packages: py4j, python-dotenv, pyparsing, pydantic, psutil, packaging, cryptography, marshmallow, environs, iso15118
  Attempting uninstall: cryptography
    Found existing installation: cryptography 38.0.4
    Uninstalling cryptography-38.0.4:
      Successfully uninstalled cryptography-38.0.4
Successfully installed cryptography-38.0.1 environs-9.5.0 iso15118-0.14.1 marshmallow-3.19.0 packaging-21.3 psutil-5.9.4 py4j-0.10.9.7 pydantic-1.10.2 pyparsing-3.0.9 python-dotenv-0.21.0
debian@BeagleBone:~/iso15118-master$ make run-secc
/usr/bin/python iso15118/secc/main.py
Traceback (most recent call last):
  File "/home/debian/iso15118-master/iso15118/secc/main.py", line 6, in <module>
    from iso15118.secc.controller.simulator import SimEVSEController
  File "/home/debian/.local/lib/python3.9/site-packages/iso15118/secc/controller/simulator.py", line 11, in <module>
    from aiofile import async_open
ModuleNotFoundError: No module named 'aiofile'
make: *** [Makefile:88: run-secc] Error 1
debian@BeagleBone:~/iso15118-master$

shalinnijel2, 2022-11-30:
Nice - this looks better - would you like to try running `pip install aiofile` and retry?

hughsheehy, 2022-11-30:
Yep. That worked. Runs exactly as it did with the poetry install.  Same command sequence as above and it ran. And I gotta tell you, installing poetry on a small device is a big effort. 

I do get the same error I'm trying to figure out on this other issue. Running on Beagle Bones turns out to be fiddly!

https://github.com/SwitchEV/iso15118/issues/171

closed 2022-12-15 by hughsheehy