ecog-io/iso15118 #171: Runs fine on Raspberry Pi.  Runs but fails on Beaglebone Black

hughsheehy, 2022-11-30:
Not sure if this is a bug or an issue or what. 

We've managed to install the library on a Raspberry Pi, and it works fine. Car talks to charger and vice versa. All works! Excellent!

But on a BeagleBone Black, the conversation starts up but then fails.  A few of us have trawled through the logs, but really have no idea what might be happenning. 

I've pasted the logs below....a bit long, but maybe someone can see something. 

H

+++++++++++++++++++++++++++++++++++++++++++

**ISO 15118 Dialog on BeagleBone**

### _+++++++++++++++++EVSE Side++++++++++++++++++++++_
(iso15118-py3.9) debian@BeagleBone:~/iso15118-master$ make run-secc
make[1]: Entering directory '/home/debian/iso15118-master'
/home/debian/.cache/pypoetry/virtualenvs/iso15118-wboF4ZSG-py3.9/bin/python iso15118/secc/main.py
DEBUG 2022-11-30 10:27:00,433 - asyncio (59): Using selector: EpollSelector
INFO 2022-11-30 10:27:00,466 - iso15118.secc.secc_settings (117): SECC settings:
INFO 2022-11-30 10:27:00,468 - iso15118.secc.secc_settings (119): PKI_PATH : /home/debian/iso15118-master/iso15118/shared/pki
INFO 2022-11-30 10:27:00,471 - iso15118.secc.secc_settings (119): ENABLE_TLS_1_3 : False
INFO 2022-11-30 10:27:00,473 - iso15118.secc.secc_settings (119): MESSAGE_LOG_EXI : False
INFO 2022-11-30 10:27:00,475 - iso15118.secc.secc_settings (119): MESSAGE_LOG_JSON : True
INFO 2022-11-30 10:27:00,478 - iso15118.secc.secc_settings (119): V20_SERVICE_CONFIG : /home/debian/iso15118-master/iso15118/shared/examples/15118_20_evse_service_config.json
INFO 2022-11-30 10:27:00,488 - iso15118.secc.secc_settings (121): PROTOCOLS : ['DIN_SPEC_70121', 'ISO_15118_2', 'ISO_15118_20_AC', 'ISO_15118_20_DC']
INFO 2022-11-30 10:27:00,491 - iso15118.secc.secc_settings (121): FREE_CERT_INSTALL_SERVICE : True
INFO 2022-11-30 10:27:00,493 - iso15118.secc.secc_settings (121): STANDBY_ALLOWED : False
INFO 2022-11-30 10:27:00,495 - iso15118.secc.secc_settings (121): USE_CPO_BACKEND : False
INFO 2022-11-30 10:27:00,498 - iso15118.secc.secc_settings (121): ALLOW_CERT_INSTALL_SERVICE : True
INFO 2022-11-30 10:27:00,500 - iso15118.secc.secc_settings (121): LOG_LEVEL : DEBUG
INFO 2022-11-30 10:27:00,503 - iso15118.secc.secc_settings (121): SECC_ENFORCE_TLS : True
INFO 2022-11-30 10:27:00,505 - iso15118.secc.secc_settings (121): AUTH_MODES : ['EIM', 'PNC']
INFO 2022-11-30 10:27:00,507 - iso15118.secc.secc_settings (121): FREE_CHARGING_SERVICE : False
INFO 2022-11-30 10:27:00,510 - iso15118.secc.secc_settings (121): NETWORK_INTERFACE : eth0
DEBUG 2022-11-30 10:27:00,612 - iso15118.secc.controller.simulator (201): New Status: starting
INFO 2022-11-30 10:27:34,474 - iso15118.shared.exi_codec (174): EXI Codec version: 1.55
INFO 2022-11-30 10:27:34,478 - iso15118.secc (31): Starting 15118 version: 0.14.1
INFO 2022-11-30 10:27:34,482 - iso15118.secc.comm_session_handler (217): Communication session handler started
INFO 2022-11-30 10:27:34,517 - iso15118.secc.transport.udp_server (130): UDP server socket ready
INFO 2022-11-30 10:27:34,522 - iso15118.secc.transport.udp_server (116): UDP server started at address FF02::1%eth0 and port 15118
INFO 2022-11-30 10:27:34,527 - iso15118.secc.transport.tcp_server (109): TLS server started at address fe80::3608:e1ff:fed7:84a7%eth0 and port 64787
DEBUG 2022-11-30 10:27:34,533 - iso15118.secc.controller.simulator (201): New Status: ready
DEBUG 2022-11-30 11:01:33,579 - iso15118.secc.transport.udp_server (145): Message received from ('fe80::3608:e1ff:fed7:84a7', 41911, 0, 2): 01fe9000000000020000
INFO 2022-11-30 11:01:33,588 - iso15118.secc.comm_session_handler (328): SDPRequest received: [Security: TLS, Protocol: TCP]
INFO 2022-11-30 11:01:33,594 - iso15118.secc.comm_session_handler (368): Sending SDPResponse: [ IP address: fe80::3608:e1ff:fed7:84a7, Port: 64787 , Security: TLS , Transport: TCP ]
INFO 2022-11-30 11:01:33,681 - iso15118.secc.comm_session_handler (263): TCP client connected, client address is ('fe80::3608:e1ff:fed7:84a7', 39778, 0, 2).
INFO 2022-11-30 11:01:33,840 - iso15118.shared.comm_session (344): Starting a new communication session
INFO 2022-11-30 11:01:33,847 - iso15118.shared.states (137): Entered state SupportedAppProtocol
DEBUG 2022-11-30 11:01:33,850 - iso15118.shared.states (141): Waiting for up to 20.0 s
INFO 2022-11-30 11:01:37,290 - iso15118.shared.exi_codec (299): Decoded message (ns=urn:iso:15118:2:2010:AppProtocol): {"supportedAppProtocolReq":{"AppProtocol":[{"ProtocolNamespace":"urn:iso:15118:2:2013:MsgDef","VersionNumberMajor":2,"VersionNumberMinor":0,"SchemaID":1,"Priority":1},{"ProtocolNamespace":"urn:iso:std:iso:15118:-20:AC","VersionNumberMajor":1,"VersionNumberMinor":0,"SchemaID":2,"Priority":2}]}}
INFO 2022-11-30 11:01:37,295 - iso15118.shared.comm_session (234): supportedAppProtocolReq received
INFO 2022-11-30 11:01:37,303 - iso15118.shared.exi_codec (245): Message to encode (ns=urn:iso:15118:2:2010:AppProtocol): {"supportedAppProtocolRes": {"ResponseCode": "OK_SuccessfulNegotiation", "SchemaID": 1}}
INFO 2022-11-30 11:01:37,877 - iso15118.secc.states.sap_states (146): Chosen protocol: ISO_15118_2
INFO 2022-11-30 11:01:37,880 - iso15118.shared.comm_session (420): Sending supportedAppProtocolRes
INFO 2022-11-30 11:01:37,886 - iso15118.shared.states (137): Entered state SessionSetup
DEBUG 2022-11-30 11:01:37,888 - iso15118.shared.states (141): Waiting for up to 20.0 s
INFO 2022-11-30 11:01:41,823 - iso15118.shared.comm_session (390): The data link will terminate in 2 seconds and the TCP connection will close in 5 seconds.
INFO 2022-11-30 11:01:41,831 - iso15118.shared.comm_session (394): Reason: TCP peer closed connection
INFO 2022-11-30 11:01:43,842 - iso15118.shared.comm_session (401): terminated the data link
INFO 2022-11-30 11:01:46,847 - iso15118.shared.comm_session (408): TCP connection closed to peer with address None

### _ ++++++++++++EV Side++++++++++++++++++++++++++++++++_
debian@BeagleBone:~/iso15118-master$ make run-evcc
make[1]: Entering directory '/home/debian/iso15118-master'
/home/debian/.cache/pypoetry/virtualenvs/iso15118-wboF4ZSG-py3.9/bin/python iso15118/evcc/main.py
DEBUG 2022-11-30 11:01:01,548 - asyncio (59): Using selector: EpollSelector
INFO 2022-11-30 11:01:33,387 - iso15118.evcc.evcc_settings (93): EVCC settings:
INFO 2022-11-30 11:01:33,391 - iso15118.evcc.evcc_settings (95): PKI_PATH : /home/debian/iso15118-master/iso15118/shared/pki
INFO 2022-11-30 11:01:33,394 - iso15118.evcc.evcc_settings (95): ENABLE_TLS_1_3 : False
[... 14 lines cut ...]
INFO 2022-11-30 11:01:33,470 - iso15118.evcc.comm_session_handler (380): Remaining SDP requests: 50
DEBUG 2022-11-30 11:01:33,473 - iso15118.evcc.transport.udp_client (90): UDP client socket ready
INFO 2022-11-30 11:01:33,575 - iso15118.evcc.comm_session_handler (317): Sending SDPRequest: [Security: TLS, Protocol: TCP]
DEBUG 2022-11-30 11:01:33,584 - iso15118.evcc.transport.udp_client (170): Message sent: [Header = [0x1, 0xfe, 0x9000, 2], Payload = 0000)]
INFO 2022-11-30 11:01:33,598 - iso15118.evcc.transport.udp_client (105): Received datagram from UDP server at address ('fe80::3608:e1ff:fed7:84a7', 15118, 0, 2)
INFO 2022-11-30 11:01:33,603 - iso15118.evcc.comm_session_handler (462): SDPResponse received: [ IP address: fe80::3608:e1ff:fed7:84a7, Port: 64787 , Security: TLS , Transport: TCP ]
INFO 2022-11-30 11:01:33,606 - iso15118.evcc.comm_session_handler (398): Starting TLS client, trying to connect to fe80::3608:e1ff:fed7:84a7 at port 64787 ...
INFO 2022-11-30 11:01:33,678 - iso15118.evcc.comm_session_handler (405): TCP client connected
INFO 2022-11-30 11:01:33,844 - iso15118.shared.comm_session (344): Starting a new communication session
INFO 2022-11-30 11:01:33,854 - iso15118.shared.states (137): Entered state SupportedAppProtocol
DEBUG 2022-11-30 11:01:33,861 - iso15118.shared.states (141): Waiting for up to 2.0 s
WARNING 2022-11-30 11:01:33,864 - iso15118.evcc.comm_session_handler (160): Removed DIN_SPEC from the list of supported Protocols as TLS is enabled
INFO 2022-11-30 11:01:33,874 - iso15118.shared.exi_codec (245): Message to encode (ns=urn:iso:15118:2:2010:AppProtocol): {"supportedAppProtocolReq": {"AppProtocol": [{"ProtocolNamespace": "urn:iso:15118:2:2013:MsgDef", "VersionNumberMajor": 2, "VersionNumberMinor": 0, "SchemaID": 1, "Priority": 1}, {"ProtocolNamespace": "urn:iso:std:iso:15118:-20:AC", "VersionNumberMajor": 1, "VersionNumberMinor": 0, "SchemaID": 2, "Priority": 2}]}}
[... cut ...]
  File "/home/debian/iso15118-master/iso15118/evcc/comm_session_handler.py", line 542, in get_from_rcv_queue
    await self.restart_sdp(True)
  File "/home/debian/iso15118-master/iso15118/evcc/comm_session_handler.py", line 366, in restart_sdp
    raise SDPFailedError(
iso15118.shared.exceptions.SDPFailedError: EVCC tried to initiate a V2GCommunicationSession, but maximum number of SDP retry cycles (1) is now reached. Shutting down high-level communication. Unplug and plug in the cable again if you want to start anew.

shalinnijel2, 2022-11-30:
From a quick look at the log:
On the EVSE side:

```
INFO 2022-11-30 11:01:37,290 - iso15118.shared.exi_codec (299): Decoded message (ns=urn:iso:15118:2:2010:AppProtocol): {"supportedAppProtocolReq":{"AppProtocol":[{"ProtocolNamespace":"urn:iso:15118:2:2013:MsgDef","VersionNumberMajor":2,"VersionNumberMinor":0,"SchemaID":1,"Priority":1},{"ProtocolNamespace":"urn:iso:std:iso:15118:-20:AC","VersionNumberMajor":1,"VersionNumberMinor":0,"SchemaID":2,"Priority":2}]}}
INFO 2022-11-30 11:01:37,295 - iso15118.shared.comm_session (234): supportedAppProtocolReq received
INFO 2022-11-30 11:01:37,303 - iso15118.shared.exi_codec (245): Message to encode (ns=urn:iso:15118:2:2010:AppProtocol): {"supportedAppProtocolRes": {"ResponseCode": "OK_SuccessfulNegotiation", "SchemaID": 1}}
INFO 2022-11-30 11:01:37,877 - iso15118.secc.states.sap_states (146): Chosen protocol: ISO_15118_2
INFO 2022-11-30 11:01:37,880 - iso15118.shared.comm_session (420): Sending supportedAppProtocolRes
```
EVSE seems to have responded within ~600ms of receiving supportedAppProtocolReq from the EV - which looks good.

However on the EV side, a timeout error is thrown after 2 seconds after waiting for supportedAppProtocolRes. Timeout for supportedAppProtocolReq is 2 seconds. 
```
INFO 2022-11-30 11:01:34,784 - iso15118.shared.comm_session (420): Sending supportedAppProtocolReq
INFO 2022-11-30 11:01:36,799 - iso15118.shared.comm_session (390): The data link will terminate in 2 seconds and the TCP connection will close in 5 seconds.
INFO 2022-11-30 11:01:36,801 - iso15118.shared.comm_session (394): Reason: TimeoutError occurred. Waited for 2.0 s
```

Maybe worth checking if there are any services on the EV side if there are any other services running. Also maybe worth checking on the EVSE side perhaps the pcap file to see if it was indeed sent out within 600ms?

hughsheehy, 2022-11-30:
Will do. 

I wonder if it's also a possibiilty that the poor Beagle Bone Black just isn't able to run both the SECC and the EV code at the same time without so much latency that it times out. Poor thing isn't very fast. 

Would it/might it be possible to run - for instance - the SECC code on the Beagle Bone and the EV code on a Pi if they're on the same network and subnet?

hughsheehy, 2022-12-01:
So, interesting.  And puzzling. 

I'm running the code on a Beagle Bone and on a Raspberry Pi, both connected to the same network. 
If I run the Pi as the EV and the BeagleBone as the SECC, I get timeout errors. (see  [link](https://docs.google.com/document/d/1FxeNmcqtHsgYSQVvv5mpnRLEWP7TWdJ7QHs4Cwdxe3U/edit?usp=sharing) )

If I run the Pi as the SECC and the BeagleBone as the EV, it works fine. Or at least it goes to the CurrentDemandReq and gets a CurrentDemandRes and then repeats. But it does that much.   (see  [link   ](https://docs.google.com/document/d/1AijZDIF_Hr30Wo3nTBdrXBZuOb0RJhiDN0xGQCq8HBU/edit?usp=sharing))

I have TLS turned off in the  .env file on both sides.

hughsheehy, 2022-12-01:
It looks as if the BeagleBone is taking too long to respond to the first "supportedAppProtocolReq".  As in it takes three seconds to even decode the message, and in the meantime the EV side (which will only wait 2.0 seconds) has given up and closed the TCP connection. 

Now a BeagleBone isn't a fast machine, but it seems odd that decoding such a short message takes three seconds. That feels as if it might be a bug in the code that's holding things up somewhere.   A Pi is a lot faster and if the code is running on any other machine then the problem might not be noticed at all.

hughsheehy, 2022-12-01:
I changed all the 2.0 second timeouts in the various files to be 20 seconds instead. Then ran everything again. 

It all worked until it got to one where the timeout limit was still set at 5.0 seconds.  

But looking at the logs, the BeagleBone takes up to 10 seconds to respond to requests from the EV side.  That seems a long time for such a response..

++++++++++++++++++++++++++++++++++++++++++++++++++++

INFO    2022-12-01 13:13:22,017 - iso15118.shared.exi_codec (299): Decoded message (ns=urn:iso:15118:2:2010:AppProtocol): {"supportedAppProtocolRes":{"ResponseCode":"OK_SuccessfulNegotiation","SchemaID":1}}
INFO    2022-12-01 13:13:22,018 - iso15118.shared.comm_session (231): supportedAppProtocolRes received
INFO    2022-12-01 13:13:22,020 - iso15118.evcc.states.sap_states (142): Chosen protocol: ISO_15118_2
INFO    2022-12-01 13:13:22,021 - iso15118.shared.exi_codec (245): Message to encode (ns=urn:iso:15118:2:2013:MsgDef): {"V2G_Message": {"Header": {"SessionID": "00"}, "Body": {"SessionSetupReq": {"EVCCID": "E45F01B2CCCB"}}}}
INFO    2022-12-01 13:13:22,372 - iso15118.shared.comm_session (415): Sending SessionSetupReq
INFO    2022-12-01 13:13:22,373 - iso15118.shared.states (137): Entered state SessionSetup
DEBUG    2022-12-01 13:13:22,373 - iso15118.shared.states (141): Waiting for up to 20.0 s
INFO    2022-12-01 13:13:32,889 - iso15118.shared.exi_codec (299): Decoded message (ns=urn:iso:15118:2:2013:MsgDef): {"V2G_Message":{"Header":{"SessionID":"3D1DADCFE788C382"},"Body":{"SessionSetupRes":{"ResponseCode":"OK_NewSessionEstablished","EVSEID":"UK123E1234","EVSETimeStamp":1669900407}}}}
INFO    2022-12-01 13:13:32,890 - iso15118.shared.comm_session (231): SessionSetupRes received
INFO    2022-12-01 13:13:32,891 - iso15118.shared.exi_codec (245): Message to encode (ns=urn:iso:15118:2:2013:MsgDef): {"V2G_Message": {"Header": {"SessionID": "3D1DADCFE788C382"}, "Body": {"ServiceDiscoveryReq": {}}}}
INFO    2022-12-01 13:13:33,134 - iso15118.shared.comm_session (415): Sending ServiceDiscoveryReq
INFO    2022-12-01 13:13:33,135 - iso15118.shared.states (137): Entered state ServiceDiscovery
DEBUG    2022-12-01 13:13:33,135 - iso15118.shared.states (141): Waiting for up to 20.0 s

hughsheehy, 2022-12-15:
So - on this - it seems the Beagle Bone is just too slow to run the Java EXI codec.

closed 2022-12-15 by hughsheehy

tropxy, 2022-12-19:
Hi Hugh,

Yeah, the Java EXI codec is a real bottleneck for embedded devices, that is why in the josev_pro offer we use the Rust EXI codec instead.