ecog-io/iso15118 #200: Questions about the operating environment of this project?

Baymax-Rob, 2023-02-20:
I want to know if this project can run on ARM architecture platform? If not, which platform is recommended to run on?

shalinnijel2, 2023-03-13:
Hi,

Yes, it works on arm architecture. Our commercial offering - Josev Pro - is supported on armv7, armv8 and linux x86_64 environments.

closed 2023-03-13 by shalinnijel2

Baymax-Rob, 2023-04-07:
I am very interested in the ISO15118 charging protocol, and thank you very much for your reply, as well as your organization's open source ISO15118 project source code. At present, I am actively trying to port this part of the code to the development board of AMV7, but I have encountered some problems, I hope you can give me some pointers, as follows:
1. After the SLAC sequence matching of ISO15118-3 is completed, what actions need to be taken to send the V2G Message of ISO15118-2? For example, after the SLAC matching is completed, do I need to send any important parameters?
2. Does the ISO15118-Python project run on the ARMV7 development board necessarily require poetry, a packaging dependency management tool, to run the code normally? (PS: Because I use less python, it is limited to being able to understand)