ecog-io/iso15118 #212: Sequence Error after Welding Detection

jpo-stx, 2023-03-20:
There is an issue with the welding detection processing:

- EV sends the first WeldingDetectionReq with 'Ongoing'.
- EVSE responds with WeldingDetectionRes with it's present voltage.
- EV sends SessionStopRes
 
This is not correct according to the specification:

**[V2G20-1630]**  After receiving the DC_WeldingDetectionReq with EVProcessing set to ‘Ongoing’, the SECC shall  respond  with  a  DC_WeldingDetectionRes  within V2G_SECC_Msg_Performance_Time  according  to  Table  217.  The  next  allowed  request shall  be  DC_WeldingDetectionReq  and  the  V2G_SECC_Sequence_Timeout  is  set according to Table 217. 

**[V2G20-1631]**  After receiving the DC_WeldingDetectionReq with EVProcessing set to ‘Finished’, the SECC shall  respond  with  a  DC_WeldingDetectionRes  within V2G_SECC_Msg_Performance_Time  according  to  Table  217.  The  next  allowed  request shall  be  SessionStopReq  and  the  V2G_SECC_Sequence_Timeout  is  set  according  to Table 215. 

As long as a WeldingDetectionReq with 'Ongoing' is sent the next request message has to be a WeldingDetctionReq

jpo-stx, 2023-03-20:
Thank you for the promt response!

I already implemented a workaround. I can create a pull request.

I think it only affects the ISO15118-20. In the ISO15118-2 and DIN70121-2 protocol the EV can send the SessionStopReq after the first WeldingDetectionRes was received, because there is no processing parameter in the "old" protocols.

tropxy, 2023-03-22:
Yes, we would be very happy with your contribution @jpo-stx