ecog-io/iso15118 #216: PROTOCOLS=ISO_15118_20_AC doesn't work normally

GUANMINLIAO, 2023-03-23:
Hello:
          When I type PROTOCOLS=ISO_15118_20_AC in .env file, it's doesn't work normally.Does this problem need me to adjust any settings?Or is it a bug? Thank you!

tropxy, 2023-03-23:
Hi, can you include the message logs so we can inspect the issue?

GUANMINLIAO, 2023-04-13:
Hi @tropxy 

Sorry,it is late for response.
First,I type "PROTOCOLS=ISO_15118_20_AC" in .env file.
[image]

Then, this is SECC log:
[image]

and EVCC log is below:
[image]

Could you please tell me if there is any method to slove this issue, Many thanks!

shalinnijel2, 2023-04-13:
The failure happened during supportedAppProtocol request - even though the secc is configured to offer -20, it seems the EV requested only -2. From the above .env file, EVCC_CONFIG_PATH is pointing at the sample -2 config. Could you try updating it as follows and retry?
```
EVCC_CONFIG_PATH=iso15118/shared/examples/evcc/iso15118_20/evcc_config_ac_bpt.json
```

GUANMINLIAO, 2023-04-13:
Hi @shalinnijel2 

It runs normally on the shell. Many thanks for the help!

closed 2023-04-13 by GUANMINLIAO