ecog-io/iso15118 #220: data link and Tcp connection get closed early in process

prcpratik, 2023-04-04:
i established a ethernet connection between my laptop and rpi . 

started to simulate evcc code on laptop and secc code on rpi .

both are connected on same ethernet connection with ip assigned to laptop and dhcp server assign ip rpi.

connection is established and message exchange started but after some time connection get closed due to some error.

shalinnijel2, 2023-04-13:
From the logs, it seems like the supportedAppProtocolRes that was sent by SECC was never received and hence a timeout error was thrown at the EVCC end. Is this behaviour consistent? This does look strange - a connection was indeed established. Outbound tcp traffic from EVCC looks fine as well, but inbound was n't received. Any suspicious firewall settings on the EVCC side?

prcpratik, 2023-04-26:
yes ,issue is resolved .i mistakenly added more delay on in shared.states code .now it is working fine