ecog-io/iso15118 #270: Runing DC in 0.20.0

GUANMINLIAO, 2023-07-05:
Could I ask a problem about this?

How do I set in the .env file to run the iso15118-20 dc ?
The following is my .env file running iso15118-20 ac:
[image]
Many thanks!

Baymax-Rob, 2023-07-05:
In the readme of this project：Enabled communication protocols on EVCC. NOTE: ISO 15118 DC support is still under development.

shalinnijel2, 2023-07-05:
Two changes:
1. To support secc to offer 15118-20 DC, set:
```
PROTOCOLS=ISO_15118_20_DC
```
2. To set evcc to request dc, set evcc to use the following example config:
```
EVCC_CONFIG=iso15118/shared/examples/evcc/iso15118_20/evcc_config_dc_bpt.json
```
@Baymax-Rob Thank you for pointing that out. Will update the readme. DC support was added some time ago. This comment was missed.

closed 2023-07-05 by GUANMINLIAO