ecog-io/iso15118 #280: EXIDecodingError occurred while processing message  in state AuthorizationSetup

shubh2016shiv, 2023-07-23:
While using ISO15118_20 protocol using CG5317 hardware, the code is breaking in authorization step on SECC side. 
However, when I run the communication protocol for EVCC and SECC on same local machine, the code runs fine. 

Below is the error which I am getting from the communication session logs:
iso15118.shared.comm_session (221): EXIDecodingError (Exception): Transformer Exception: java.io.EOFException: Premature EOS found while reading data.

Is there any way to handle it or bypass it.

shalinnijel2, 2023-07-24:
I wonder if the message to be decoded has been passed to the codec at all. 
Could you share the AuthorizationSetup message and the corresponding exi stream, please?

SebaLukas, 2023-08-15:
Hi :)

i have the same problem that josev can not decode the lower exi stream. Same error message as described above.

Exi stream: `800c04790c8adceeee09688d6cac3a6062000200`
AuthorizationSetupRes:
```
{
    "AuthorizationSetupRes": {
        "Header": {"SessionID": "F21915B9DDDC12D1", "TimeStamp": 1691411798},
        "ResponseCode": "OK",
        "AuthorizationServices": ["EIM"],
        "CertificateInstallationService": true,
        "EIM_ASResAuthorizationMode": {}
    }
}
```

shalinnijel2, 2024-01-15:
Hi @SebaLukas ...closing this issue following your comment.

closed 2024-01-15 by shalinnijel2