ecog-io/iso15118 #290: The vehicle battery SOC is missing on iso15118-2 AC charging.

coollhc77, 2023-08-17:
Is there a way or option to send the vehicle battery SOC?

INFO    2023-08-14 11:55:00,759 - iso15118.shared.exi_codec (301): Decoded message (ns=urn:iso:15118:2:2013:MsgDef): {"V2G_Message":{"Header":{"SessionID":"3F8ABB0C6EE88C2F"},"Body":{"ChargingStatusReq":{}}}}
INFO    2023-08-14 11:55:00,760 - iso15118.shared.comm_session (243): ChargingStatusReq received
INFO    2023-08-14 11:55:00,762 - iso15118.shared.exi_codec (245): Message to encode (ns=urn:iso:15118:2:2013:MsgDef): {"V2G_Message": {"Header": {"SessionID": "3F8ABB0C6EE88C2F"}, "Body": {"ChargingStatusRes": {"ResponseCode": "OK", "EVSEID": "UK123E1234", "SAScheduleTupleID": 1, "ReceiptRequired": false, "AC_EVSEStatus": {"NotificationMaxDelay": 0, "EVSENotification": "None", "RCD": false}}}}}
INFO    2023-08-14 11:55:00,927 - iso15118.shared.comm_session (436): Sent ChargingStatusRes

tropxy, 2023-08-20:
Hi @coollhc77, unfortunately in AC charging, the ISO 15118-2 does not foresee an option for the EV to send its SoC or battery capacity. The only EV data you get is the one during the charge parameter Discovery:
[image]
Unfortunately, this means that if you need that data, you need to get it externally, like via the EV telematics API

closed 2023-08-20 by tropxy

coollhc77, 2023-08-20:
Thanks for your detailed explanation.