ecog-io/iso15118 #295: ISO15118-2 TLS fails.

coollhc77, 2023-09-08:
Is there a problem with my setup?

**SECC Logs.**
DEBUG    2023-09-08 17:49:28,463 - asyncio (59): Using selector: EpollSelector
INFO    2023-09-08 17:49:28,471 - iso15118.secc.secc_settings (110): SECC settings:
INFO    2023-09-08 17:49:28,472 - iso15118.secc.secc_settings (112): ENABLE_TLS_1_3                : True
INFO    2023-09-08 17:49:28,472 - iso15118.secc.secc_settings (112): V20_SERVICE_CONFIG            : /home/coollhc/iso15118/Josev/iso15118/iso15118/shared/examples/secc/15118_20/service_config.json
INFO    2023-09-08 17:49:28,472 - iso15118.secc.secc_settings (112): MESSAGE_LOG_EXI               : True
INFO    2023-09-08 17:49:28,472 - iso15118.secc.secc_settings (112): MESSAGE_LOG_JSON              : True
INFO    2023-09-08 17:49:28,472 - iso15118.secc.secc_settings (112): PKI_PATH                      : /home/coollhc/iso15118/Josev/iso15118/iso15118/shared/pki
INFO    2023-09-08 17:49:28,472 - iso15118.secc.secc_settings (114): USE_CPO_BACKEND               : False
INFO    2023-09-08 17:49:28,472 - iso15118.secc.secc_settings (114): AUTH_MODES                    : ['EIM', 'PNC']
INFO    2023-09-08 17:49:28,473 - iso15118.secc.secc_settings (114): LOG_LEVEL                     : DEBUG
INFO    2023-09-08 17:49:28,473 - iso15118.secc.secc_settings (114): NETWORK_INTERFACE             : eth0
INFO    2023-09-08 17:49:28,473 - iso15118.secc.secc_settings (114): FREE_CHARGING_SERVICE         : False
INFO    2023-09-08 17:49:28,473 - iso15118.secc.secc_settings (114): SECC_ENFORCE_TLS              : True
INFO    2023-09-08 17:49:28,473 - iso15118.secc.secc_settings (114): PROTOCOLS                     : ['ISO_15118_20_AC']
INFO    2023-09-08 17:49:28,473 - iso15118.secc.secc_settings (114): ALLOW_CERT_INSTALL_SERVICE    : True
INFO    2023-09-08 17:49:28,473 - iso15118.secc.secc_settings (114): FREE_CERT_INSTALL_SERVICE     : True
INFO    2023-09-08 17:49:28,473 - iso15118.secc.secc_settings (114): STANDBY_ALLOWED               : False
DEBUG    2023-09-08 17:49:28,482 - iso15118.secc.controller.simulator (213): New Status: starting
INFO    2023-09-08 17:49:32,448 - iso15118.shared.exi_codec (174): EXI Codec version: 1.55
INFO    2023-09-08 17:49:32,449 - iso15118.secc (31): Starting 15118 version: 0.23.2
INFO    2023-09-08 17:49:32,449 - iso15118.secc.comm_session_handler (229): Communication session handler started
INFO    2023-09-08 17:49:32,451 - iso15118.secc.transport.udp_server (134): UDP server socket ready
INFO    2023-09-08 17:49:32,452 - iso15118.secc.transport.udp_server (120): UDP server started at address FF02::1%eth0 and port 15118
DEBUG    2023-09-08 17:49:32,462 - iso15118.secc.controller.simulator (213): New Status: ready
DEBUG    2023-09-08 17:49:44,166 - iso15118.secc.transport.udp_server (155): Message received from ('fe80::6930:7013:d79e:ea2a', 42813, 0, 2): 01fe9000000000020000
INFO    2023-09-08 17:49:44,168 - iso15118.secc.comm_session_handler (401): SDPRequest received: [Security: TLS, Protocol: TCP]
INFO    2023-09-08 17:49:44,191 - iso15118.secc.transport.tcp_server (134): TLS server started at address fe80::603c:7c4c:5fb4:ae20%eth0 and port 60758
INFO    2023-09-08 17:49:44,200 - iso15118.secc.comm_session_handler (428): Sending SDPResponse: [ IP address: fe80::603c:7c4c:5fb4:ae20, Port: 60758 , Security: TLS , Transport: TCP ]

**EVCC Logs.**
DEBUG    2023-09-08 17:49:39,734 - asyncio (59): Using selector: EpollSelector
INFO    2023-09-08 17:49:39,742 - iso15118.evcc.evcc_settings (45): EVCC environment settings:
INFO    2023-09-08 17:49:39,742 - iso15118.evcc.evcc_settings (47): ENABLE_TLS_1_3                : True
INFO    2023-09-08 17:49:39,742 - iso15118.evcc.evcc_settings (47): MESSAGE_LOG_JSON              : True
INFO    2023-09-08 17:49:39,742 - iso15118.evcc.evcc_settings (47): V20_SERVICE_CONFIG            : /home/coollhc/iso15118/Josev/iso15118/iso15118/shared/examples/secc/15118_20/service_config.json
INFO    2023-09-08 17:49:39,742 - iso15118.evcc.evcc_settings (47): MESSAGE_LOG_EXI               : True
INFO    2023-09-08 17:49:39,742 - iso15118.evcc.evcc_settings (47): PKI_PATH                      : /home/coollhc/iso15118/Josev/iso15118/iso15118/shared/pki
INFO    2023-09-08 17:49:39,744 - iso15118.evcc.evcc_settings (49): EVCC_CONFIG_PATH              : iso15118/shared/examples/evcc/iso15118_2/evcc_config_pnc_ac.json
INFO    2023-09-08 17:49:39,744 - iso15118.evcc.evcc_settings (49): NETWORK_INTERFACE             : eth0
INFO    2023-09-08 17:49:39,744 - iso15118.evcc.evcc_settings (49): LOG_LEVEL                     : DEBUG
INFO    2023-09-08 17:49:39,749 - iso15118.evcc.evcc_config (116): EVCC Settings
INFO    2023-09-08 17:49:39,749 - iso15118.evcc.evcc_config (119): supported_energy_services     : [<ServiceV20.AC: 1>]
INFO    2023-09-08 17:49:39,750 - iso15118.evcc.evcc_config (119): is_cert_install_needed        : True
INFO    2023-09-08 17:49:39,750 - iso15118.evcc.evcc_config (119): use_tls                       : True
INFO    2023-09-08 17:49:39,750 - iso15118.evcc.evcc_config (119): sdp_retry_cycles              : 1
INFO    2023-09-08 17:49:39,750 - iso15118.evcc.evcc_config (119): max_contract_certs            : 3
INFO    2023-09-08 17:49:39,750 - iso15118.evcc.evcc_config (119): enforce_tls                   : False
INFO    2023-09-08 17:49:39,750 - iso15118.evcc.evcc_config (119): supported_protocols           : [<Protocol.ISO_15118_2: (<Namespace.ISO_V2_MSG_DEF: 'urn:iso:15118:2:2013:MsgDef'>, <enum 'ISOV2PayloadTypes'>)>]
INFO    2023-09-08 17:49:39,750 - iso15118.evcc.evcc_config (119): energy_transfer_mode          : AC_three_phase_core
INFO    2023-09-08 17:49:39,751 - iso15118.evcc.evcc_config (119): max_supporting_points         : 1024
INFO    2023-09-08 17:49:44,065 - iso15118.shared.exi_codec (174): EXI Codec version: 1.55
INFO    2023-09-08 17:49:44,065 - iso15118.evcc (30): Starting 15118 version: 0.23.2
INFO    2023-09-08 17:49:44,066 - iso15118.evcc.comm_session_handler (307): Communication session handler started
DEBUG    2023-09-08 17:49:44,067 - iso15118.evcc.comm_session_handler (386): Initiating new SDP cycle, 0 more cycles(s) left
INFO    2023-09-08 17:49:44,067 - iso15118.evcc.comm_session_handler (392): Remaining SDP requests: 50
DEBUG    2023-09-08 17:49:44,068 - iso15118.evcc.transport.udp_client (90): UDP client socket ready
INFO    2023-09-08 17:49:44,169 - iso15118.evcc.comm_session_handler (329): Sending SDPRequest: [Security: TLS, Protocol: TCP]
DEBUG    2023-09-08 17:49:44,169 - iso15118.evcc.transport.udp_client (170): Message sent: [Header = [0x1, 0xfe, 0x9000, 2], Payload = 0000)]
INFO    2023-09-08 17:49:44,206 - iso15118.evcc.transport.udp_client (105): Received datagram from UDP server at address ('fe80::603c:7c4c:5fb4:ae20', 15118, 0, 2)
INFO    2023-09-08 17:49:44,207 - iso15118.evcc.comm_session_handler (475): SDPResponse received: [ IP address: fe80::603c:7c4c:5fb4:ae20, Port: 60758 , Security: TLS , Transport: TCP ]
INFO    2023-09-08 17:49:44,207 - iso15118.evcc.comm_session_handler (410): Starting TLS client, trying to connect to fe80::603c:7c4c:5fb4:ae20 at port 60758 ...
ERROR    2023-09-08 17:49:44,224 - iso15118.evcc.comm_session_handler (419): SSLCertVerificationError when trying to connect to host fe80::603c:7c4c:5fb4:ae20 and port 60758
Traceback (most recent call last):
  File "/home/coollhc/iso15118/Josev/iso15118/iso15118/evcc/comm_session_handler.py", line 414, in start_comm_session
    self.tcp_client = await TCPClient.create(
  File "/home/coollhc/iso15118/Josev/iso15118/iso15118/evcc/transport/tcp_client.py", line 57, in create
    raise exc
  File "/home/coollhc/iso15118/Josev/iso15118/iso15118/evcc/transport/tcp_client.py", line 48, in create
    self.reader, self.writer = await asyncio.open_connection(
  File "/usr/lib/python3.9/asyncio/streams.py", line 52, in open_connection
    transport, _ = await loop.create_connection(
  File "/usr/lib/python3.9/asyncio/base_events.py", line 1081, in create_connection
    transport, protocol = await self._create_connection_transport(
  File "/usr/lib/python3.9/asyncio/base_events.py", line 1111, in _create_connection_transport
    await waiter
  File "/usr/lib/python3.9/asyncio/sslproto.py", line 528, in data_received
    ssldata, appdata = self._sslpipe.feed_ssldata(data)
  File "/usr/lib/python3.9/asyncio/sslproto.py", line 188, in feed_ssldata
    self._sslobj.do_handshake()
  File "/usr/lib/python3.9/ssl.py", line 944, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate signature failure (_ssl.c:1123)

Baymax-Rob, 2023-09-11:
First, please confirm that you generated the certificate as follows:
#### 1. Generate certificates

The project includes a script to help on the generation of -2 and -20 certificates.
This script is located under `iso15118/shared/pki/` directory and is called `create_certs.sh`.
The following command provides a guide for the script usage:

```bash
$ cd iso15118/shared/pki/
$ ./create_certs.sh -h
```

Use the following commands to generate certificates for ISO 15118-2 and 15118-20:

```bash
$ ./create_certs.sh -v iso-2
$ ./create_certs.sh -v iso-20
```
Then, if your EVCC and SECC are run separately on two computers, then you must copy the EVCC pki folder to the SECC computer (that is, ensure that the certificates generated by the above operations must be consistent); if so If you run EVCC and SECC locally at the same time, you need to check the time validity of the certificate.

coollhc77, 2023-09-11:
Copying the EVCC pki to the SECC is missing.
The CERTIFICATE_VERIFY_FAILED issue has been resolved.

Thank you Baymax-Rob

closed 2023-09-11 by coollhc77