ecog-io/iso15118 #326: Unable to set ECDH-ECDSA cipher suite on client (EVCC) side

anudeep-20, 2023-11-23:
https://github.com/SwitchEV/iso15118/blob/5e627d335575ebf85f1ef6350a835115f12ca058/iso15118/shared/security.py#L201

When changing the EVCC supported cipher suite as `ECDH-ECDSA-AES128-SHA256` instead of `ECDHE-ECDSA-AES128-SHA256`, getting the following SSL error.
 `ssl.SSLError: ('No cipher can be selected.',)` 

Please let me know, how this can be modified or any alternate ways to make this modification.

longrujun, 2024-01-06:
Python uses OpenSSL. OpenSSL does not support ECDH-ECDSA-AES128-SHA256 in the new version.

kaabia, 2024-05-25:
The Python `ssl` module is typically linked against your system's OpenSSL library when you install it. To check your current OpenSSL version, run the `openssl version` command.

New OpenSSL versions do not support certain ciphers like ECDH. If you need support for these ciphers, you'll need an older OpenSSL version. I recommend using [OpenSSL 1.0.2j](https://ftp.openssl.org/source/old/1.0.2/openssl-1.0.2j.tar.gz).

To compile Python with OpenSSL 1.0.2j, follow these steps:

1. **Download, compile, and install OpenSSL 1.0.2j**:

```
~$ wget https://ftp.openssl.org/source/old/1.0.2/openssl-1.0.2j.tar.gz
~$ tar xvf openssl-1.0.2j.tar.gz
~$ cd openssl-1.0.2j
~$ ./config --prefix=/usr/local/ssl --openssldir=/usr/local/ssl
~$ make
~$ sudo make install
```

After installation, run `/usr/local/ssl/bin/openssl ciphers` to verify that the list of supported ciphers includes `ECDH-ECDSA-AES128-SHA256`.

2. **Download, configure, and compile Python 3.9.1**:

2.1. Download step:

```
~$ wget https://www.python.org/ftp/python/3.9.1/Python-3.9.1.tar.xz
~$ tar xvf Python-3.9.1.tar.xz
~$ cd Python-3.9.1
```

2.2. Configuration step:

Open the `Setup` file located in the `Python-3.9.1/Modules` directory and uncomment the following lines:

```
# Socket module helper for SSL support; you must comment out the other
# socket line above, and possibly edit the SSL variable:
SSL=/usr/local/ssl
_ssl _ssl.c \
    -DUSE_SSL -I$(SSL)/include -I$(SSL)/include/openssl \
    -L$(SSL)/lib -lssl -lcrypto
```

Next, run the `configure` script with the appropriate flags:

```
~/Python-3.9.1$ ./configure --enable-optimizations --with-openssl=/usr/local/ssl
```

2.3. Compilation step:

```
~/Python-3.9.1$ make
```

3. **Verify the OpenSSL version used by Python**:

After the compilation is successful, you can verify that Python is using the desired OpenSSL version (1.0.2j) by running the following commands:

```
~/Python-3.9.1$ ./python
Python 3.9.1 (default, May 25 2024, 22:41:22)
[GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ssl
>>> ssl.OPENSSL_VERSION
'OpenSSL 1.0.2j  26 Sep 2016'
>>>
```

By following these steps, you'll compile Python 3.9.1 and link it against the OpenSSL 1.0.2j version, which supports the desired ciphers like `ECDH-ECDSA-AES128-SHA256`.

kaabia, 2024-05-25:
Now if you want to change the Python version used by your Poetry project to the custom Python 3.9.1 installation compiled with OpenSSL 1.0.2j. consider the following steps:

1. **Verify the Python installation path**: Make sure that the path `/home/mr_bean/Python-3.9.1/python` is correct and points to the Python 3.9.1 installation you compiled with OpenSSL 1.0.2j.

2. **Use the `env` command instead of `whereis`**: The `whereis` command may not always provide the correct path to the Python executable used by Poetry. It's better to use the `env` command to get the path to the Python executable used by the current Poetry environment:

```
~/workspace/iso15118$ poetry env info
```

This command will display the path to the Python executable used by the current Poetry environment.

3. You can use the `poetry env use` command to switch the Python version used by your Poetry environment. This command ensures that Poetry updates its environment correctly:

```
poetry env use /home/mr_bean/Python-3.9.1/python
```

After running this command, Poetry will use the specified Python installation for the current project.

4. **Update the `pyproject.toml` file**: It's recommended to update the `python` dependency constraint in the `pyproject.toml` file to match the version you're using. For example:

```toml
[tool.poetry.dependencies]
python = "^3.9.1"
```

This will ensure that Poetry installs dependencies compatible with Python 3.9.1.

5. **Reinstall dependencies**: After changing the Python version, you may need to reinstall the project dependencies to ensure they are compatible with the new Python version:

```
poetry install
```

By following these steps, you should be able to use the custom Python 3.9.1 installation compiled with OpenSSL 1.0.2j for your Poetry project. However, keep in mind that using a custom Python installation may introduce compatibility issues with some dependencies or tools that expect a system-provided Python installation.