ecog-io/iso15118 #355: Is there a more detailed manual

minner2, 2023-12-31:
I am a novice and I want to learn the relevant knowledge of iso 15118-20. May I ask the author if there are more detailed usage documents for this project？

shalinnijel2, 2024-01-09:
These are a few resources:
https://youtu.be/ztwnmyya-nQ?t=311
https://www.switch-ev.com/blog/switch-to-clarity-whats-new-in-iso-15118-20

Regarding getting the project up and running, could you try the instructions here:
https://github.com/SwitchEV/iso15118/tree/375cd0bb66505321ac4390578f051138575161ec?tab=readme-ov-file#local-installation
SECC env settings are listed here:
https://github.com/SwitchEV/iso15118/tree/375cd0bb66505321ac4390578f051138575161ec?tab=readme-ov-file#environment-settings

Feel free to reply to this thread or get in touch directly if anything isn't clear and we will help you get up and running.

shalinnijel2, 2024-01-15:
Hi @minner2 - I will close this ticket for now - but feel free to reopen and add more questions here.

closed 2024-01-15 by shalinnijel2