ecog-io/iso15118 #356: evse_notification  incorrectly used to detect shutdown request from EVSE in dinspec

heavyweight87, 2024-01-04:
Hi,

For din spec on the EV side, EVSE initated shutdown is checked like this:

`if dc_evse_status.evse_notification == EVSENotification.STOP_CHARGING:
    logger.debug("EVSE Notification received requesting to stop charging.")
    await self.stop_charging()`

DIN spec states however that evse_notification should always be set to None.
What you actually want to check is DCEVSEStatusCode.

There for if the EVSE wants to initate shutdown it currently will not work

shalinnijel2, 2024-01-09:
Hi @heavyweight87 ,
Thank you for pointing this out. Linking the corresponding requirement below:

```
[V2G-DC-500] For DC charging according to DIN SPEC 70121, the value of EVSENotification shall always be set to “None”.
```
And the following regarding EVSEStatusCode:

```
In the context of DIN SPEC 70121, “EVSE_Shutdown” is used by the SECC to trigger a termination of the charging process in situations other than an emergency shutdown, see chapter 9.7.4.2.2. “EVSE_EmergencyShutdown” may be used by the SECC to inform the EVCC that the EVSE is in the process of, or has just executed, an emergency shutdown. However, to force the EV to participate in the emergency shutdown, the SECC does not use this value, but instead uses the CPL in accordance with IEC 61851-23. In all other cases, the expected value for EVSEStatusCode is “EVSE_Ready” or “EVSE_IsolationMonitoringActive”.
```

I have created an internal ticket to fix this. Will update here once it goes in to a release. Or feel free to open a PR to fix this if this is quite urgent for you and we can merge the fix.

heavyweight87, 2024-01-21:
Hi
I opened a PR for this as well as a previous issue I opened about the EVCC not having a way to indicate when charging is enabled (i.e. going to state C) #368

PR #368 (2024-01-23): EVCC enable charging and fix for DIN EVSE initiated stop
Fixes #356 and #264 

- Added enable_charging to evcc controller interface. This gives the EVCC the opertunity to go to state C (or B)
- Fixed an issue where the wrong flag was checked when receiving current demand response for EVSE initiated shutdown (DIN only)
closed 2024-01-23 by shalinnijel2