ecog-io/iso15118 #357: EVCC keeps alive after charging session finishes

touchlinux, 2024-01-08:
I am using `iso15118/evcc` to test `custom secc` developed in-house. It is basically replacing `iso15118/secc` to our secc. The expectation was to keep the secc running while reconfigure and re-run evcc over time to see if our secc supports various protocols and use cases.
However, there is a problem that EVCC does not stop even after finishing the charging session. It keeps running until it gets exception to stop. I think it would be more graceful if evcc terminates when it gets `StopNotification` for some cases including mine.
I could add new bool configuration in EVCC_CONFIG file, to specify the EVCC behavior whether it should be alive or terminated after a charging session.
[0001-is_alive_after_session.patch](https://github.com/SwitchEV/iso15118/files/13856593/0001-is_alive_after_session.patch)

shalinnijel2, 2024-01-09:
Hi @touchlinux, 
Thank you for this. Could you open a PR for this? We will give it a test and then get it merged?
Kind regards,
Shalin.

touchlinux, 2024-01-10:
Thank you for feedback @shalinnijel2.
Unfortunately, I could not push my branch to upstream.
It seems I have a permission issue. Can you help me?
```
$ git push --set-upstream origin imp/is_alive_after_session
Username for 'https://github.com': touchlinux
Password for 'https://touchlinux@github.com': 
remote: Permission to SwitchEV/iso15118.git denied to touchlinux.
fatal: unable to access 'https://github.com/SwitchEV/iso15118.git/': The requested URL returned error: 403
```

shalinnijel2, 2024-01-10:
Ah, sounds like you forgot to fork the repo before pushing the branch?

touchlinux, 2024-01-10:
@shalinnijel2 PR is created. Can you review it?
https://github.com/SwitchEV/iso15118/pull/361

shalinnijel2, 2024-01-22:
Closing as this as the above [PR](https://github.com/SwitchEV/iso15118/pull/361) now merged to master.

closed 2024-01-22 by shalinnijel2